solarmd-pty-ltd/myPower24-WEB #10: Micasa project - Update 3 - Logger Map

Kyle-Swanepoel, 2020-10-19:
## Proposed portal enhancements

The logger map is also enhanced to include more detail.  Each pin on the map now includes the additional [dashboard](https://github.com/solarmd-pty-ltd/myPower24-WEB/issues/8) data as specified above (Load, Grid import, Total consumption, Battery %, Active alerts, Load shedding status).  A new feature is to include a RED pin for offline loggers and to reallocate the orange pins to a new function.

- Green pin - Online logger, no active alerts
- Orange pin - Online logger with active alerts
- Red pin - Offline logger

The map should also include a feature to hide / show loggers by status.

https://github.com/solarmd-pty-ltd/myPower24-WEB/projects/2#column-11276887

[image]

platar86, 2020-11-08:
What would be the definition of Loadshedding Status?

tiaanv, 2020-11-08:
GRID Voltage 0V, or perhaps Inverter warning MAINS VOLTAGE error?

tiaanv, 2020-11-08:
Loadshedding Status was not part of our original requirement:
[image]

tiaanv, 2021-05-19:
progress?

closed 2024-01-15 by KaloyanDi