solarmd-pty-ltd/myPower24-WEB #11: Micasa project - Update 4 - Logger events

Kyle-Swanepoel, 2020-10-19:
## Proposed portal enhancements

The events tab now includes a read and unread function as well as a filtering capability.  The inclusion of the emailing events to designated email addresses is important (please confirm if this function is working as it should).

- An addition for an event for Low Battery - generated when the battery has discharged to the designated value.

https://github.com/solarmd-pty-ltd/myPower24-WEB/projects/2#column-11276887

[image]

merlobranco, 2021-01-27:
Already implemented, please take a look to the attached screenshots
[image]
[image]

tiaanv, 2021-01-27:
@merlobranco batter low event? event when battery falls below configured threshold?

merlobranco, 2021-01-27:
Still working on that

tiaanv, 2021-05-19:
progress?

closed 2024-01-15 by KaloyanDi