solarmd-pty-ltd/myPower24-WEB #13: Micasa project - Update 6 - Logger general settings

Kyle-Swanepoel, 2020-10-19:
## Proposed portal enhancements

The inclusion of 2 new features under the General settings tab.

- Plant sqm - To be entered by the client and used for additional reporting calculations (average consumption per sqm).
- Active check box - Automatically ticked for new loggers but can be unticked for decommissioned loggers and forms part of the filtering in the Logger overview.
 
https://github.com/solarmd-pty-ltd/myPower24-WEB/projects/2#column-11276887

[image]

Masud-Sonday, 2021-01-26:
@tiaanv Please specify what the ''Active'' check box does.
More precisely please advise what the ''Non active'' state of the button would do in terms of the logger history and visibility.

tiaanv, 2021-01-26:
@Masud-Sonday. Active is an indicator for reporting purposes.  There are a number of "Consolidated"  reports/"views"(pages). This flag simply includes/excludes the logger from these views/reports.
@Kyle-Swanepoel please share the document that I sent, as well as your interpretations with @Masud-Sonday for insight.  I have attached my original document for reference, and Kyle's response doc
[Logger Portal Enhancements technical feedback 13-10-2020.docx](https://github.com/solarmd-pty-ltd/myPower24-WEB/files/5873326/Logger.Portal.Enhancements.technical.feedback.13-10-2020.docx)
[Interconnect Functional Specification Portal Enchancements V1.0.docx](https://github.com/solarmd-pty-ltd/myPower24-WEB/files/5873327/Interconnect.Functional.Specification.Portal.Enchancements.V1.0.docx)

Kyle-Swanepoel, 2021-01-26:
@Masud-Sonday The active function unticked would be to decommission a logger.  If the logger were to be replaced.  It would assist in filtering loggers that are in use and those that aren't.  History data would remain as the logger still remains viewable if required.

Kyle-Swanepoel, 2021-01-26:
https://docs.google.com/document/d/1-4JU_562y7O5xjFoXGB-qqBJ3aJKa-4pqhkshN2Tw2I/edit?usp=sharing

@Masud-Sonday this was the shared document that everyone has access to.
The Github repositories are my responses to the brief based on our offering at the time.

tiaanv, 2021-05-19:
progress?

closed 2024-01-15 by KaloyanDi