solarmd-pty-ltd/myPower24-WEB #8: Micasa project - Update 1 - Logger overview (Logger list)

Kyle-Swanepoel, 2020-10-19:
## Proposed portal enhancements

The main purpose of this page will be to incorporate the majority of the proposed changes into our existing logger view and provide them with an overview of their sites at a glance.  The original view becomes condensed with an addition of 6 columns displaying data at a predetermined refresh rate.  This includes the following:

- Consumption / Load
- Grid import
- Consumption Total
- Battery %
- Active Alerts / Events
- Load shedding status

In addition to this a filtering option is available on all columns.  Inclusive in filtering by column is the option to view "Active" and "Inactive" loggers which is initially set under the General settings tab.

https://github.com/solarmd-pty-ltd/myPower24-WEB/projects/2#card-47503070

[image]

Masud-Sonday, 2021-01-26:
@tiaanv we already created a plant as an object that can group single or multiple loggers and display the data together. What are your thoughts?

tiaanv, 2021-01-26:
@Masud-Sonday These are very much individual plants. They are plotted independently as well on the map.  I'm not saying it can't work, from a reporting perspective, but I am not sure how this works in practice.  We can perhaps have an interactive session to go through how this could work, based on the requirements?

tiaanv, 2021-05-19:
Can we have a session discussing this?

closed 2024-01-15 by KaloyanDi