solarmd-pty-ltd/myPower24-WEB #9: Micasa project - Update 2 - Logger options (Logger list)

Kyle-Swanepoel, 2020-10-19:
## Proposed portal enhancements

Under this view, settings remain much the same apart from the inclusion of 2 new features.

- History data - Takes the user directly to the History data tab of the logger.  A large feature for the client is the reporting side and having quick access to this is important. 
- General settings - Easily accessible for the client to update.

https://github.com/solarmd-pty-ltd/myPower24-WEB/projects/2#column-11276887

[image]

Masud-Sonday, 2021-01-26:
@tiaanv We prefer to keep the contents menu simple and small for the purpose of viewing on smaller devices such as mobile.
Please consider using the the ''Set as current'' button, then navigating to the required page.

tiaanv, 2021-01-26:
@Masud-Sonday. I can accept this as per your comment, and since I have been using the portal now, I also agree with this approach. I actually cannot recall this specifically in our requirements to be in this format.

merlobranco, 2021-01-27:
@Masud-Sonday @platar86 as discussed yesterday the following direct links where in deviceMonitor and deviceSearch views were created
[image]

closed 2021-01-27 by merlobranco