solarmd-pty-ltd/myPower24-WEB #20: Override not clearing under certain battery conditions

Masud18650, 2020-10-22:
BMS firmware version: 130.
Bootloader version: 103.

Condition: Battery has error 17 (cell overvoltage) and error 29 (open cell detected, due to 1 very unbalanced cell)

Issue:  When  overriding ON for 2 seconds, the BMS closes the relay as expected, however after the 2 sec has elapsed the override status persists, until you manually clear it. The same is true for override OFF.

Notes: Once battery recovered from all errors, the test was performed again and override function worked correctly

[image]

platar86, 2021-11-19:
To be tested in fw166

platar86, 2021-11-19:
This issue can not be recreated with normal overvoltage undervoltage error. There is bug fix in the code but it must be revalidated. ver166

closed 2021-11-19 by platar86