solarmd-pty-ltd/myPower24-WEB #21: Sub device model filter

Masud18650, 2020-10-29:
Sub device model filter allows you to check multiple boxes but only applies the filter to the last selected sub device model.

[image]

merlobranco, 2020-11-30:
The affirmation of tis issue is not accurate, what it does the filter is providing the devices which has an ABR_TRIO_50 OR an AXPERT_KING OR an Axpert, etc.

In the following example, taken from http://login.mypower24.co.za/

We can see we have: 

> 0 loggers with ABB_TRIO_50

[image]

> 81 loggers with AXPERT_KING

[image]

> 716 loggers with Axpert

[image]

> 794 loggers with ABB_TRIO_50 OR AXPERT_KING OR  Axpert. 794 loggers that have ANY of the selected sub device models

[image]

If the affirmation of this issue were right the filter should return 716 loggers NOT 794 loggers

merlobranco, 2020-11-30:
I could add an checkbox when is clicked is giving the logger that have ALL the sub devices selected.
So we could retrieve the loggers that have ANY of the selected sub device models OR 
retrieve the loggers that have ALL the selected sub device models

Masud18650, 2020-11-30:
this would be preferably, thanks

merlobranco, 2021-01-13:
Already implemented

closed 2021-01-13 by merlobranco