solarmd-pty-ltd/myPower24-WEB #22: Carlo Gavazzi ET340 meter Instant Values Total not updating

tiaanv, 2020-11-08:
Meter Total Values not updating
[image]

platar86, 2020-11-08:
Logger serial number please

tiaanv, 2020-11-08:
SLV201554232 | Demo unit ( Old Mutual) Bayside Mall

platar86, 2020-11-08:
this is preliminary update, we do not include in this versions caching files update. please clear your browser cache manually.

platar86, 2020-11-08:
Going to remove energy counter for Apparent power and reactive for individual phases since they are calculated. 
[image]

tiaanv, 2020-11-08:
> this is preliminary update, we do not include in this versions caching files update. please clear your browser cache manually.

Clearing browser cache had no impact on TOTAL VALUES updating.  Individual Phases seem OK form start.

platar86, 2020-11-08:
please verify 
[image]

tiaanv, 2020-11-08:
My Issue is referring to instant values:
[image]

tiaanv, 2020-11-26:
I am not sure if this is completed yet, but I have seen correct values at least from FW 7.120

Kyle-Swanepoel, 2021-01-26:
Hi Tiaan,

Currently values are reporting as follows.  Please confirm you are happy?

[image]

tiaanv, 2021-01-26:
close please

closed 2021-01-26 by Kyle-Swanepoel