solarmd-pty-ltd/myPower24-WEB #23: Logger Page Keeps loading energy meter modal box

tiaanv, 2020-11-08:
SERIAL:         SLV201554232
FIRMWARE:   7.103

=========================

Navigating to My Devices-> Logger

Brings up Meter as modal, and no other functions can be accessed
Close Button also disabled
[image]

platar86, 2020-11-08:
@Masud18650 please rise this for high priority for Monday. @tiaanv please do soft reboot to fix it temporary.

tiaanv, 2020-11-08:
> @Masud18650 please rise this for high priority for Monday. @tiaanv please do soft reboot to fix it temporary.

Will do Reboot,

tiaanv, 2020-11-20:
This is a more serious problem if we have no local access to device, since rebooting remotely becomes a problem

Masud18650, 2020-11-20:
Issue resolved in logger update version: 7.120

closed 2020-11-20 by Masud18650