solarmd-pty-ltd/myPower24-WEB #29: How to include 3ph Meter (ET340) data on Dashboard

tiaanv, 2020-11-08:
SERIAL: SLV201554232
FIRMWARE: 7.103
=========================

Linking Total of 3phase meter does not work.
There is an exisiting Power Calculator that calculates Total Energy from the three individual phases (I did not create it).
Linking the "Total Energy" node to  the "GRID OUT" Node. Renders no Data on the dashboard.

The expected result is that the INSTANT POWER, and TOTAL Energy values are used and displayed on the dashboard for the Utility/Grid Feed:
[image]

tiaanv, 2020-11-09:
:)

I saw.. Great!
[image]

tiaanv, 2020-11-09:
I can see this was achieved by linking all three phases to the GRID OUT node....  
[image]

Is the way it will always be done, or will we eventually have a Node representing the "INSTANT VALUES TOTAL" from the CV 3ph meter?  This makes more sense to me.
[image]

platar86, 2020-11-12:
In progress ...

tiaanv, 2020-11-26:
> In progress ...

I can already see that the Meter Value Total is updating
[image]

I can also see a new node under power type indicating Total power.  However, linking it to Grid out does not render values on the dashboard.  I know I am jumping the gun. Not expecting it to be complete.  Just giving running feedback.

Kyle-Swanepoel, 2021-01-26:
Hi Tiaan,

Current progress - Power type calculations are reflecting correctly on the dashboard.  The 3 phases are added together under a new type (Total Power).

[image]

[image]

platar86, 2021-10-07:
@Kyle-Swanepoel , please close id sorted