solarmd-pty-ltd/myPower24-WEB #31: Unable to set Plant Address

tiaanv, 2020-11-09:
SERIAL: N/A
FIRMWARE: VARIOUS
=========================
The following dialog is shown indicating google maps API issue
[image]
When searching for a location, no matter what is entered the following dialog is shown
[image]
The map itself shows "for development purposes only"
[image]

tiaanv, 2020-11-09:
related to #33 
This is now working.

Kyle-Swanepoel, 2020-11-09:
* Address map working.

[image]

* Full overview logger map working

[image]

closed 2020-11-10 by tiaanv