solarmd-pty-ltd/myPower24-WEB #32: Logger Map not working (specific user)

tiaanv, 2020-11-09:
When logging in as the following user:

OMF | omfportal@micasa.co.za | 11/05/2020 16:05:40by: SolarMD_Support | Enabled | 18 | Super user

The Loggers Map fails to load/Display at all.

This could be due to no valid GPS data for the loggers (related #31 )

tiaanv, 2020-11-09:
Seems that when the loggers have invalid address/GPS details the Map fails to load.... I updated all the loggers in the profile, and the map loaded successfully.

platar86, 2020-11-09:
@Kyle-Swanepoel , please do not ignore this issue!

Kyle-Swanepoel, 2020-11-09:
* Logger with incorrect address - defaults to current address (Map functional).

[image]

***

* Address option pull through when typing the name.

[image]

***

* Correct address 

[image]

***

Potential bug - Does not force valid address to be entered but does require a character input to save.

tiaanv, 2020-11-26:
When setting up site General details for the first time, the map does not work, due to no entry in the "Installation address" entering any value in there does not follow the normal search operations, and does not update the map.  However.  Typing in Anything.  Then refreshing the page will allow for normal functionality (expected behavior) to resume.  Most likely just requires an elegant catch for initial null value.

BrilliesNate, 2021-11-19:
HI @Kyle-Swanepoel  as discussed, all features are fixed and working.

closed 2021-11-19 by BrilliesNate