solarmd-pty-ltd/myPower24-WEB #34: Axpert Ai data not displaying

Kyle-Swanepoel, 2020-11-10:
We are having reports from clients not being able to see any data under Axpert Ai on mobile devices.  

All users have been advised to clear their browser cache and cookies.  Browsers affected include Google Chrome, Firefox and Safari.

Issue on Google Chrome includes latest version available on play store - 86.0.4240.185.

Devices include iPhone, iPad, and Android devices.

Kyle-Swanepoel, 2020-11-19:
Update:

The issue seems to be affecting iPhone and Mac users on all browsers.  These users are unable to view any data on mypower24.

Kyle-Swanepoel, 2020-12-02:
Issue seems to be resolved on all other devices apart from Apple devices.

henry-thomas, 2020-12-09:
Should be solved in next update. Have a working version in development.

@Kyle-Swanepoel, if you hear about any other issues regarding apple users then please let us know so we can investigate.

Kyle-Swanepoel, 2020-12-09:
@henry-thomas fantastic.  Thank you.  Will keep you posted on any other related issues.

closed 2021-01-26 by henry-thomas