solarmd-pty-ltd/myPower24-WEB #39: Logger - Static IP

SipheloSkepe, 2020-11-24:
Hi Team, 

We are requiring the functionality to set a static IP address for the network adapters on the logger. Currently this is not working.

This also forms part of a requirement for the Micasa project due to the restrictive nature of the networks in the banking sector.

Kyle-Swanepoel, 2020-11-24:
Issue opened by Kyle Swanepoel

Kyle-Swanepoel, 2020-11-25:
[image]

Functionality is working.

Steps required to set static IP for Eth0 interface.

Connect to logger locally via WiFi or Eth1>
My Devices>
Logger mypower>
Network>
Eth0>
TCP/IP ADDRESS:  > Static >
TCP IP Info > Static > Enter IP (Adapter static configuration) >
Save >

[image]

To confirm after reboot, connect Ethernet cable to Eth0 to see network configuration.

closed 2020-11-25 by Kyle-Swanepoel