solarmd-pty-ltd/myPower24-WEB #42: SMA SunnyIsland Bridge - Ping missing - v7.122

Kyle-Swanepoel, 2020-12-09:
Hi team,

We have an issue with SMA SunnyIsland bridge on the loggers.  Ping to the SI is missing and causing a discrepancy between the data we are sending and the data we are receiving from the SI.  We have an increase in the number of support calls for batteries over or under discharging. 

A reboot of the logger temporarily fixes the issue, but the ping soon disappears.  It seems that some of the sites are even working on Lead Acid Emulation, but this was not a change made by us.

[image]
[image]
[image]

LeonidasHanos-debug, 2020-12-14:
Hi team, 

The main issue is that the SI are going into a fault state due to battery under/over voltage errors, there are no corresponding errors on mypower24.

There currently are 3 x sites that are reporting the same issue.

SLV200956309 - Bridging technology
SLV201502890 - Kobus
SLV209980540 - Material boards main
SLV210690625 - Material boards extension cluster
SLV212286807 - Leonhard Eins 
SLV210010492 - Leonhard Eins
SLV207265502 - Leonhard Eins

[image]

Masud-Sonday, 2021-01-26:
Ping missing fixed in version 7.1241 logger update.

Kyle-Swanepoel, 2021-01-28:
@Masud-Sonday Hi Mas,

We have clients reporting daily that their SMA sites are switching off due to Battery under voltage readings on SI.  There are still problems with Nkau, please see excerpt from Sunny Portal.

[image]

Further issues on this off grid site: (Access has been granted on portal - Plaas Krige)
[image]
Waiting on 2nd logger to update to 7.1241
[image]
[image]
[image]

## At the moment, logger soft reboot is still switching the SI off on most sites.

Masud-Sonday, 2021-02-01:
@Kyle-Swanepoel Sunny island issues fixed in logger update version: 7.191
- Ping Missing
- 6609 Battery Under voltage
- 9331 Meas. range violation battery voltage in device at L1

closed 2021-02-03 by Masud-Sonday