solarmd-pty-ltd/myPower24-WEB #44: Axpert King - PV not displaying

Kyle-Swanepoel, 2021-01-13:
Hi team,

PV value not displaying for Axpert King on the dashboard.  PV is also noted as "false" under OPERATING on the inverter.

Please look into this.

Thanks

[image]
[image]
[image]
[image]

platar86, 2021-09-15:
@Masud-Sonday you can test this with our UPS inverter, please propose solution based on protocol instructions or calculation.

henry-thomas, 2021-09-17:
Fixed with logger version above 8.220

closed 2021-09-17 by henry-thomas