solarmd-pty-ltd/myPower24-WEB #51: Logger - AEG inverter integration

Kyle-Swanepoel, 2021-02-18:
Hi team,

This request is from Herman Redelinghuys regarding integration of the AEG inverters via MODBUS to the logger.  I've attached the MODBUS protocol documents sent to me by the client.

I believe we have an AEG inverter in the store that we could potentially use for testing.

The client is requesting features much like the Axpert Ai interface if possible.
- Load shedding prep function
- Bypass/Error bypass mode to grid 
- Ability to turn charge controller ON/OFF (if possible)
- Time of use table - Refer to Issue #50 for a similar spec.  Client wants to do load shifting and peak shaving.

Their intention is also to make use of the logger relays, I believe the current expressions will be more than sufficient for their use case.

Thanks,
Kyle
[8000048225_05_BAL_EN (1).pdf](https://github.com/solarmd-pty-ltd/myPower24-WEB/files/6000390/8000048225_05_BAL_EN.1.pdf)
[Convert SC Flex_BAL_Modbus RTU_TCP_EN (1).pdf](https://github.com/solarmd-pty-ltd/myPower24-WEB/files/6000391/Convert.SC.Flex_BAL_Modbus.RTU_TCP_EN.1.pdf)

Khykn, 2021-05-24:
Communication established with the inverters.  Both inverters have static IP addresses.  Modbus TCP available.

Port 3 on switch - *75.120 - Battery Charger
Port 4 on switch - *75.121 - Convert SC Flex
port - 502

[image]

Kyle-Swanepoel, 2021-08-10:
Integration complete
***
[image]

closed 2021-08-10 by Kyle-Swanepoel