solarmd-pty-ltd/myPower24-WEB #53: Logger server disconnects

Kyle-Swanepoel, 2021-04-06:
Hi team,

We have multiple users who are experiencing periodic daily logger disconnects to the mypower24 server.  Some of them have replaced their loggers and are still experiencing issues.  One user has done some extensive digger and we require some assistance on this please. (SLV202103730) - New logger        
(SLV207747191) - Old logger

Can you please investigate.

Thanks

______________________________________________________________________________________________________________________________________

[04/06, 12:18] SolarMD Support: Thirdly, my logger is till going online/offline despite that it was replaced by a brand new logger. And before you ask, the logger is not on WiFi, it's connected with a CAT5 cable. On the weekend I have done the following:
1. Reset the logger
2. Reset my Internet router
3. Tested with a second CAT5 cable that is working
The logger keeps going online/offline - You can see that from the video
[04/06, 12:18] SolarMD Support: Whoever is on shift tonight and is unable to answer these 4 questions, then I want this to be escalated tomorrow to someone more senior that are able to answer me. 
I would also like a call back on 082 446 2670 so that person can explain to me what should be done next. I have had it with these one liner responses on WhatsApp.
[04/06, 12:18] SolarMD Support: Fourthly: I want to log a ticket with my ISP to see if they are dropping or blocking any traffic to . I want to know the following please:
1. What are your public facing IP addresses - It looks like it's 197.242.147.240?
2. What ports should be open - Is it both 80 and 443?

C:\Users\Johan>nslookup mypower24.dedicated.co.za
Non-authoritative answer:
Name:    mypower24.dedicated.co.za
Address:  197.242.147.240
[04/06, 12:18] SolarMD Support: I want answer on all 4 my questions please. 

The quality of support that I have received thus far after spending R133,000 is appalling to say the least. I know the support is free, but my system is not working and I cannot manage my inverter as the dashboard and system is inaccessible.  Next steps are Hello Peter and Social Media. I don't threaten easily, but this issue with the logger is now persisting for over a year. The logger started giving issues 2 weeks after installation and we also had 4 BMS events where the power were lost entirely, as the battery went below 10% (90% is the current setting) and the system did not switch to Eskom. 

When myself or Claude are asking you questions, then the default answer is to ask the client to check hi WiFi. This BS stops tonight. I am also in IT and I am not your average user. Please do not insult me by asking me stupid questions you ask your average user that won't understand this.
[04/06, 12:18] SolarMD Support: Also, I know that you are hosting with Afrihost:

C:\Users\Johan>tracert 197.242.147.240

Tracing route to mypower24.dedicated.co.za [197.242.147.240]
over a maximum of 30 hops:

  1     1 ms    <1 ms    <1 ms  www.routerlogin.com [192.168.7.1]
  2     3 ms     2 ms     3 ms  rb-bng-2.za.mtnns.net [196.4.87.104]
  3     4 ms     3 ms     3 ms  196.44.29.97
  4     4 ms     3 ms     3 ms  196-60-9-221.ixp.joburg [196.60.9.221]
  5     4 ms     4 ms     4 ms  100.127.0.171
  6     5 ms     5 ms     4 ms  169-1-21-100.ip.afrihost.co.za [169.1.21.100]
  7     *        *        *     Request timed out.
  8     5 ms     4 ms     4 ms  mypower24.dedicated.co.za [197.242.147.240]
[04/06, 12:18] SolarMD Support: When doing tracert to 197.242.147.240, there is a timeout on hop 7, which is somewhere between afrihost and your server. Is there perhaps a firewall that blocks ICMP or is there an issue?
[04/06, 12:18] SolarMD Support: Running a continuous ping also looks fine, so I doubt if the issue is with my ISP:

C:\Users\Johan>ping 197.242.147.240 -t

Pinging 197.242.147.240 with 32 bytes of data:
Reply from 197.242.147.240: bytes=32 time=5ms TTL=57
Reply from 197.242.147.240: bytes=32 time=5ms TTL=57
//removed for conciseness 
Reply from 197.242.147.240: bytes=32 time=5ms TTL=57

Ping statistics for 197.242.147.240:
    Packets: Sent = 108, Received = 108, Lost = 0 (0% loss),
Approximate round trip times in milli-seconds:
    Minimum = 4ms, Maximum = 10ms, Average = 5ms
Control-C
^C
C:\Users\Johan>
[04/06, 12:18] SolarMD Support: Pathping shows similar timeout:

C:\Users\Johan>pathping 197.242.147.240

Tracing route to mypower24.dedicated.co.za [197.242.147.240]
over a maximum of 30 hops:
  0  DellT30 [192.168.7.3]
  1  www.routerlogin.com [192.168.7.1]
  2  rb-bng-2.za.mtnns.net [196.4.87.104]
  3  196.44.29.97
  4  196-60-9-221.ixp.joburg [196.60.9.221]
  5  100.127.0.171
  6  169-1-21-100.ip.afrihost.co.za [169.1.21.100]
  7     *        *        *
Computing statistics for 150 seconds...
            Source to Here   This Node/Link
Hop  RTT    Lost/Sent = Pct  Lost/Sent = Pct  Address
  0                                           DellT30 [192.168.7.3]
                                0/ 100 =  0%   |
  1    0ms     1/ 100 =  1%     1/ 100 =  1%  www.routerlogin.com [192.168.7.1]
                                0/ 100 =  0%   |
  2    2ms     0/ 100 =  0%     0/ 100 =  0%  rb-bng-2.za.mtnns.net [196.4.87.104]
                                0/ 100 =  0%   |
  3    3ms     0/ 100 =  0%     0/ 100 =  0%  196.44.29.97
                                0/ 100 =  0%   |
  4    4ms     0/ 100 =  0%     0/ 100 =  0%  196-60-9-221.ixp.joburg [196.60.9.221]
                                0/ 100 =  0%   |
  5  ---     100/ 100 =100%   100/ 100 =100%  100.127.0.171
                                0/ 100 =  0%   |
  6    4ms     0/ 100 =  0%     0/ 100 =  0%  169-1-21-100.ip.afrihost.co.za [169.1.21.100]

Trace complete.

platar86, 2021-04-06:
>    4. What are your public facing IP addresses - It looks like it's
>    5. What ports should be open - Is it both 80 and 443?

Please do not ever ask client
 to replace logger for disconnect issue's

Kyle-Swanepoel, 2021-04-06:
We didn't, the client replaced the logger out of their own.

Kyle-Swanepoel, 2021-04-13:
SLV215647325 - at SMD
SLV203781590 - at SMD
SLV214345108 - at SMD
SLV214677531 - at SMD

Kyle-Swanepoel, 2021-04-19:
SLV202103730 - Working
SLV214345108 - Working

closed 2021-05-04 by Masud-Sonday