solarmd-pty-ltd/myPower24-WEB #61: Client email database - Mypower24 notifications

Kyle-Swanepoel, 2021-04-29:
Hi team,

Can we please implement a feature through Mypower24 that will allow us to send a mass email to our client database.

This would be helpful for server down time notifications or upgrade features etc.

* Exclude users who are not marked as "Active"
* Create a new button that opens an email like window to send to all users
* Allow for Subject line and Body for email
 
[image]

[image]

platar86, 2021-12-30:
@henry-thomas this is already implemented in sense of asking users to revalidate their email. It must be somewhere in LogginCOntroller.java please check if everything is working correct before uploading to server.

henry-thomas, 2022-01-12:
Added Email Announcement for admin. Users need to opt in first at General Settings -> User Privacy, then they will receive the announcements. 

It also forces users to validate emails before they can opt in.

closed 2022-01-12 by henry-thomas