solarmd-pty-ltd/myPower24-WEB #62: Goodwe - B30 command - Battery force charging

LeonidasHanos-debug, 2021-06-07:
Hi Team, 

We are experiencing issues with Goodwe sites where the inverter is receiving a B30 battery force charge command.

SLV206516037 - House Shultze on SEMS portal
SLV214677531 -House Crafford on SEMS portal

@Masud-Sonday has the Goodwe protocol.

Please let us know if you require any further information.

Kyle-Swanepoel, 2021-06-18:
SLV200932377 - SunSynk reporting the same message.  Logger on 7.1933
[image]

LeonidasHanos-debug, 2021-06-28:
SLV216387005 - added 5 x sites to SMA bridge as a test.

Still force charging with either bridge.

LeonidasHanos-debug, 2021-08-10:
SLV211017486 - 3P Sunsynk installation with force charge issue.

[image]

platar86, 2021-09-15:
Test new bridge and close please. @

Kyle-Swanepoel, 2021-09-17:
Working well

***

[image]

closed 2021-09-17 by Kyle-Swanepoel