solarmd-pty-ltd/myPower24-WEB #66: SunSynk - Force charging from grid

Kyle-Swanepoel, 2021-08-10:
Sunsynk single and 3 phase inverters are force charging from grid.  The inverter is receiving a force charge command to charge the batteries.

SLV200932377 - SunSynk reporting force charge message.  Logger on 7.1933
[image]

***

SLV211017486 - 3P Sunsynk installation with force charge issue.

[image]
***

platar86, 2021-09-15:
@Kyle-Swanepoel please close this if its sorted with the new bridge.

Kyle-Swanepoel, 2021-09-17:
Working well

***

[image]

closed 2021-09-17 by Kyle-Swanepoel