solarmd-pty-ltd/myPower24-WEB #67: Dashboard values exponentially duplicated

Kyle-Swanepoel, 2021-09-02:
Dashboard values are exponentially duplicated.  The power types are duplicating themselves on the loggers.

[image]

Kyle-Swanepoel, 2021-09-02:
Current list of affected loggers:

SLV207783946 - Jan Griekwastad 
SLV210680831 - Andre HRE Tech
SLV201512555 - David Morrell
SLV206873263 - Dave Boardman
SLV208068963 - Jacques 
SLV214300598  - Werner Terblanche 
SLV211042575    John O' Sullivan
SLV204669928  - Amos pinnicle
SLV210255805 - Kevin Gallagher
SLV208068963  - Jacques 
SLV214355141 - Keegan Bentley 
SLV 201512555  - David Morrell
SLV211147554 - Kate
SLV208068963 - Jacques

platar86, 2021-09-15:
temporary fixed, please let us if there is other again. This issue is completely removed in CWEB, but we can not help with the old server.

closed 2021-09-15 by platar86