solarmd-pty-ltd/myPower24-WEB #70: Cweb - Not picking up individual inverters on Axpert Ai

Kyle-Swanepoel, 2021-09-15:
Not picking up individual inverters on Axpert AI

TypeError: Cannot read properties of undefined (reading 'eq')  On deviceConnector 1.1

[image]

***

[image]

***

What we should be seeing

[image]

platar86, 2021-09-15:
@Kyle-Swanepoel  please ask Michael to prepare one board with Voltronik in training room. They are installed already, only need cable tray and cables.

platar86, 2021-10-05:
this buttons can be replaced with Tab panel

platar86, 2021-11-18:
[image]
Fixed by someone :)

closed 2021-11-18 by platar86