solarmd-pty-ltd/myPower24-WEB #95: Login myPower -> My Devices -> Outputs -> Settings

Kyle-Swanepoel, 2021-11-12:
Styling issues on the page, not able to easily select the expression you wish to work on.

[image]

BrilliesNate, 2021-11-12:
[image]
Fixed. flex-direction column

closed 2021-11-12 by BrilliesNate

Kyle-Swanepoel, 2021-11-12:
[image]
Just an example of how it should look

BrilliesNate, 2021-11-12:
[image]