solarmd-pty-ltd/myPower24-WEB #98: Axpert Ai settings - decimal values

MilaKaloyan, 2021-11-18:
[image]

Axpert Ai > Settings > Load-shedding Mode Settings > Usage to be set as a whole number without decimal.

[image]

Axpert Ai > Settings > Settings used for Dynamic Stage adjustment> Usage to be set as a decimal number for (voltages) incl Settings for Floating stage.

MilaKaloyan, 2021-11-18:
Axpert Ai > Settings > Load-shedding Mode Settings > Usage  returns error when trying to change it

platar86, 2021-11-18:
[image]

platar86, 2021-11-18:
Decimal place issue. Fixed and uploaded.

closed 2021-11-18 by platar86