solarmd-pty-ltd/myPower24-WEB #100: Sunsynk - time of use/peak based filling device control not complete

LeonidasHanos-debug, 2021-11-23:
Hi Team, 

Our sunsynk integration is not 100% complete, the peak based filling is empty.

The system mode screen is effectively the heart of the system & the settings that most users would like to access.

[image]

Solarman App pulls the data as follows 

[image]

[image]

[image]

[image]

If you need an account to check the device control please send a message and I will provide an account.

henry-thomas, 2022-01-31:
This is ready for testing @KaloyanDi @Kyle-Swanepoel 

Can we get one Sunsynk ready for testing please?

henry-thomas, 2022-09-08:
This has been done for a while, will be in a future logger update if not there already.

closed 2022-09-08 by henry-thomas