solarmd-pty-ltd/myPower24-WEB #132: DGM Device Alert Icons

platar86, 2022-01-24:
[image]
The icons in the device panel must be the highest alert level exist, in case there is only Info alerts, icon must be same as the one in the dialog.

platar86, 2022-01-24:
also dialog title says "Alerts for",   all subdevices have field "title"

BrilliesNate, 2022-01-26:
[image]
[image]
[image]
Alerts icons fixed and display is prioritized from Error 1st then warning the info.
Title Fixed to display the device serial number.

closed 2022-01-26 by BrilliesNate