solarmd-pty-ltd/myPower24-WEB #157: 14.3kWh & Deye Controlled device Comms Failed warning

Maritha-v-Loggerenberg, 2022-11-15:
On doing direct comms with the Deye inverter, the 14.3kWh battery shows controlled device comms fail on the BMS, although the inverter picks up comms. Cables have been checked; termination switches are correct. Clients complaining about warning that does not want to clear.

BMS Hardware 301
Firmware version: 183
Inverter type: SUN-8K-SG01LP1-EU
Controlled device update delay: 100
[image]
[image]
[image]

platar86, 2022-11-25:
Hi @Maritha-v-Loggerenberg , this is inverter specific issue which we are aware of. There is a simple solution to fix it, by not listening to the inverter at all, and never throwing this warning, but then it will be more difficult for the installer to trace issues. We will put setting for coms monitor, and make it disable by default.

platar86, 2022-11-25:
Setting is available from version 185

closed 2022-11-25 by platar86

Maritha-v-Loggerenberg, 2022-11-25:
Thank you for the reply. When will version 185 be available? The clients are very unhappy with us right now.