solarmd-pty-ltd/myPower24-WEB #160: SMA- direct comms warning: "Remote device warning number 1. Specific to connected control device"

Maritha-v-Loggerenberg, 2022-11-24:
We have found a warning on a 14.3 kWh battery connected to a SMA inverter: "Remote device warning number 1. Specific to connected control device"

NOTE: we have received this warning on several SMA - 14.3kWh connections.

1. How do we help the client clear this warning in the following circumstances:
 (a) with the presence of a logger?
 (b) in absence of a logger?

2. What does this warning mean specifically?

platar86, 2022-11-25:
Hi @Maritha-v-Loggerenberg , warning 1 means that the inverter is not set for lithium battery. Please verify if this is correct with your installation.

Maritha-v-Loggerenberg, 2022-11-25:
Hi @platar86 , will have to check with the installers again.  They said they did put it on Li-mode, but we usually take that with a grain of salt.

Maritha-v-Loggerenberg, 2022-11-25:
[image]
Hi @platar86 It is indeed set to lithium mode.

platar86, 2022-11-28:
I cannot share full document with you about SMA communication because it is
confidential but small snippets:
[image: image.png]
[image: image.png]
basically, that inverter does not use external BMS information. That is why
the warning is there. I will recommend SI firmware update at least.
Is this a single case ?

Maritha-v-Loggerenberg, 2022-11-28:
Hi @platar86, not a single case no, but not a lot of people do SMA, it has become an issue with one particular site with 50 batteries connecting to SMA.

We will ask the installers to do an SI update for future sites.

closed 2024-01-16 by KaloyanDi