solarmd-pty-ltd/myPower24-WEB #165: BMU-H17 - Intermittently rebooting

Kyle-Swanepoel, 2022-12-05:
Hi,

There is still an intermittent issue with BMU-H17, fw.158 whereby the BMU reboots itself under normal operation. The SoC then halves itself.

I've also noticed this then turning the isolator ON, the BMU will reinitialize the BMSs and then close the relay.

SoC was 57% before it reset.
[image]

[image]

Kyle-Swanepoel, 2022-12-06:
BMU's reset twice this evening, SoC dropped to ~32% both times.

[image]
[image]

platar86, 2022-12-10:
[image]
none of our BMU in the factory reboots? I need more info, Site configuration and any patterns when the reboot happens. Bussy building something to catch this event in the log

Kyle-Swanepoel, 2022-12-12:
@platar86 What we have noticed is that most of the battery modules have v301 BMSs. The other sites with BMU-H17 have v304 BMSs. Can we test this in the factory on v301 BMSs?

platar86, 2023-02-17:
Please close if this issue is resolved or comment further.

Kyle-Swanepoel, 2023-02-17:
Resolved, client caused the issue by powering the logger from one of the module bus bars.

closed 2023-02-17 by Kyle-Swanepoel