solarmd-pty-ltd/myPower24-WEB #167: Wineline EV Chargers - EMS control

Kyle-Swanepoel, 2022-12-12:
Hi team,

Can we please fully integrate the EV chargers to the EMS.

If it can include all the same features as the other controlled devices.
Min power / max power limit
Min capacity / max capacity limit
TOU table

The ramp up logic also needs to be implemented properly. If we can include the Step up logic here too.

Kyle-Swanepoel, 2022-12-13:
Hi Henry, 

Seems that the chargers are either storing old logical values and reverting between the current set point and previous set points.

[image]

Kyle-Swanepoel, 2022-12-13:
If the cluster step current is below the maximum current limit for the cluster, the chargers periodically go into error state and the logic for the cluster resets and begins the current step-up from scratch.

Reducing the current step delay reduces the fluctuation in the loads and also reduces the frequency of error states on the chargers.

The videos were taken with cluster current limit of 100A, step of 100A @ 500ms
&& cluster limit 300A, step of 300A @ 500ms.

[image]

https://user-images.githubusercontent.com/72873730/207245130-d846725f-7739-416a-ab88-0de91a8d065b.mp4

https://user-images.githubusercontent.com/72873730/207245145-e55209be-464d-4906-ab95-4d3b3df2a294.mp4

Kyle-Swanepoel, 2022-12-13:
Power types are also erratic. Frequent comms loss on the power manager with the chargers. Power data jumps anywhere from 100,000W to 750,000W with a consistent load of ~240,000W on the chargers.

https://user-images.githubusercontent.com/72873730/207259648-def04acd-a9b1-40ea-ad19-a18240e62596.mp4

Kyle-Swanepoel, 2023-02-16:
> If the cluster step current is below the maximum current limit for the cluster, the chargers periodically go into error state and the logic for the cluster resets and begins the current step-up from [...]

This has been resolved. Cluster step working properly.

Kyle-Swanepoel, 2023-02-16:
> ved. Cluster step working properly.

Differences in device, logger config and mySQL database found. They are all the same now, seems to have resolved this issue.

Kyle-Swanepoel, 2023-02-16:
@henry-thomas Another issue that is prevalent at Brahman Hills. After loadshedding, when the AC supply comes back and the chargers come back on, randomly some of the chargers will report "Unbalanced current".  The only ways to fix it currently is to either drop the AC breaker to the charger and put it back on.
Or soft rebooting the Logger.

platar86, 2023-02-17:
> 

Please take this as separate issue

Kyle-Swanepoel, 2023-02-17:
> ### Power types are also erratic. Frequent comms loss on the power manager with the chargers. Power data jumps anywhere from 100,000W to 750,000W with a consistent load of ~240,000W on the chargers. >
> Please take this as separate issue

This seems to have been resolved with the cluster step issue above.

henry-thomas, 2023-02-17:
@Kyle-Swanepoel we need to reproduce what is happening at Brahman hills to see what is happening exactly. 

Forcing the logger to reboot on that error is a dirty fix, so let's not think of that as a solution. 

A - Is this the only site where this happens?
B - Do the chargers go offline when there is loadshedding?

> @henry-thomas Another issue that is prevalent at Brahman Hills. After loadshedding, when the AC supply comes back and the chargers come back on, randomly some of the chargers will report "Unbalanced [...]

Kyle-Swanepoel, 2023-02-17:
@henry-thomas I agree, rebooting the logger is not the solution to the problem.

Currently it's the only site that is having this issue with unbalanced current that I am aware of.

Yes the chargers go offline when there is loadshedding.

I am going to close this issue and open up new issues for each issue on site as this thread is getting a bit messy now.

closed 2023-02-17 by Kyle-Swanepoel