solarmd-pty-ltd/myPower24-WEB #169: Cweb issues - Jan 2023 - List

Kyle-Swanepoel, 2023-01-11:
Hi Henry,

I'm leaving a list of the issues here that we pick up as we go.

Kyle-Swanepoel, 2023-01-11:
[image]

Kyle-Swanepoel, 2023-01-11:
Advanced User > Device Search. Not working

[image]

Kyle-Swanepoel, 2023-01-11:
Admin User > Logger Updates

Updates not working for loggers, BMSs, BMUs etc.

[image]

Kyle-Swanepoel, 2023-01-11:
Sungrow Eth devices have become Solplanet

Logger myPower > Interfaces > eth0 / Ethernet local

[image]

[image]

henry-thomas, 2023-01-12:
Will get to device search, can we make that a separate issue please? 

All the rest here is done. If there is anything else, let's make them separate issues so we can close them one by one please.

closed 2023-01-12 by henry-thomas