solarmd-pty-ltd/myPower24-WEB #177: Logger V2.20 - EM series BMS - Infinity SoC

Kyle-Swanepoel, 2023-03-10:
Hi Team,

We've picked up an issue on a couple of sites with EM series BMSs where the Logger V2.20/21 is being used as the control device to the inverter.

The EM BMS's will periodically report TOTAL CYCLE: Infinity and this then get's sent to the Software Bridge and is reported as the CAPACITY value that get's sent to the Control Device.   This is causing systems to trip.

***
[image]
***
[image]

platar86, 2023-03-12:
logger bridge is Deprecated, please do not use or explain the reason why you need it.
Next to clarify something about the logger cycle, it is indeed not correct, but cycles do not get sent to the bridge at all. The bridge reports capacity infinity, which has nothing to do with cycles. Please try to convert the system to direct coms. wont have this issue

platar86, 2023-03-12:
I see, this is a mixed system with multiple batteries. Will have a look and let you know about this SF bridge issue

Kyle-Swanepoel, 2023-03-13:
@platar86 I only duplicated the Victron bridge to show the issue. The EM series batteries had to be put onto a different virtual bus to not affect the system.

Thanks