solarmd-pty-ltd/myPower24-WEB #182: "Sunsynk  Time of Use/ Peak Based Filling " buttons do not work

kamenskiq, 2023-03-28:
The weekly schedule on and off buttons do not work!
[image]

henry-thomas, 2023-03-28:
@kamenskiq Please describe the issue with some details. 

Are there error messages? Logger FW version? What are the symptoms?

henry-thomas, 2023-03-28:
What model Sunsynk is this?

kamenskiq, 2023-03-28:
No error mesages, Firmware Ver 106 , Hardware Ver 221. 
When i turn off the button and click on setting the button turns off and turns on by itself.
[image]

henry-thomas, 2023-03-28:
Can you confirm that the value is still incorrect after you refresh the page?